MikePopoloski/slang #1493: [slang][slang-tidy] Diagnostics include stack printed only once

likeamahoney, 2025-08-25:
**Is your bug report related to the C++ slang project or the pyslang Python bindings?**
C++ slang

**Describe the bug**
Suppose we have a file `source.sv`:

```verilog
`include "lib.svh"
```

and SystemVerilog header `lib.svh` with content:

```verilog
module lib_top(
    input logic a,
    inout logic b,
    output logic c
);

    reg bb;
    always_comb begin
        if (a)
            bb = c;
    end

endmodule
```

The library file intentionally violated the rules of two detectors - `STYLE-13` (`AlwaysCombBlockNamed`) and `STYLE-2` (`AlwaysCombBlockNamed`):

```
[UndrivenRange] PASS
[UnusedSensitiveSignal] PASS
[NoLegacyGenerate] PASS
[NoDotVarInPortConnection] PASS
[OnlyAssignedOnReset] PASS
[EnforceModuleInstantiationPrefix] PASS
[RegisterHasNoReset] PASS
[NoImplicitPortNameInPortConnection] PASS
[EnforcePortPrefix] PASS
[XilinxDoNotCareValues] PASS
[CastSignedIndex] PASS
[AlwaysCombBlockNamed] WARN
in file included from inc2.sv:1:
lib.svh:8:5: warning: [STYLE-13] definition of an unnamed always_comb block
    always_comb begin
    ^

[NoLatchesOnDesign] PASS
[OnlyANSIPortDecl] PASS
[NoOldAlwaysSyntax] PASS
[EnforcePortSuffix] WARN
lib.svh:2:17: warning: [STYLE-2] port 'a' is not correctly suffixed with suffix: "_i"
    input logic a,
                ^
lib.svh:3:17: warning: [STYLE-2] port 'b' is not correctly suffixed with suffix: "_io"
    inout logic b,
                ^
lib.svh:4:18: warning: [STYLE-2] port 'c' is not correctly suffixed with suffix: "_o"
    output logic c
                 ^

[AlwaysCombNonBlocking] PASS
[GenerateNamed] PASS
[AlwaysFFBlocking] PASS
[AlwaysFFAssignmentOutsideConditional] PASS
[NoDotStarInPortConnection] PASS
```

But diagnostics included stack (`in file included from inc2.sv:1`) was printed only from the first checker. Despite the fact that both detectors are triggered in the library code.  Usage of `--diag-include-stack` option doesn't change anything. 

Suppose the problem is near around this [code](https://github.com/MikePopoloski/slang/blob/master/source/diagnostics/DiagnosticEngine.cpp#L218)

Since the set of unique includes was already filled during the first checker run, no new entries were added in the second run

**To Reproduce**

```
slang-tidy source.sv
//or
slang-tidy source.sv --diag-include-stack
```

likeamahoney, 2025-08-25:
Should `slang-tidy` driver clear stack of includes or it should be managed by `DiagnosticEngine` directly?

MikePopoloski, 2025-08-26:
I suppose it's fine to clear the include stack after each check from slang-tidy.

likeamahoney, 2025-08-26:
> I suppose it's fine to clear the include stack after each check from slang-tidy.

Can this be fixed by creating a new method for DiagnosticEngine that clears the stack? Or does it require a workaround where we switch the diagnostic engine for every slang-tidy checker? Currently, there's no way to manipulate the engine stack directly because it's a private field with no related public methods.

MikePopoloski, 2025-08-26:
Yeah, a new method is fine.

closed 2025-08-27 by MikePopoloski