MikePopoloski/slang #1501: Variable definitions with parameter-based widths drop the parameter from AST

pieter3d, 2025-09-02:
```systemverilog
module test;
parameter int WIDTH=32;
logic [WIDTH-1:0] some_net;
endmodule
```
In the AST, some_net is a `VariableSymbol` with a [31:0] range. There is no way to discover if/how any parameters were involved - it appears identical to a hard-coded range. 

This is different from e.g. a `RangeSelectExpression` : `assign some_net = some_other_net[WIDTH-1:0];`. In that case the full subtraction operation with the parameter as a `NamedValue` in the tree is present as expected.

MikePopoloski, 2025-09-03:
Yes, this is true; types are uniquified and so lose some of that original syntactic information. You can always look at the syntax nodes of the variable declaration itself and bind the expression manually to see what it resolves to. Roughly:
```
auto& exprSyntax = varSymbol.getDeclaredType().getTypeSyntax().as<IntegerTypeSyntax>().dimensions[0].specifier->as<RangeDimensionSpecifierSyntax>().selector.as<RangeSelectSyntax>().left;
auto& expr = Expression::bind(exprSyntax, someContext);
```

MikePopoloski, 2025-09-04:
Where would it be retained? Types are created and compared to each other all over the place via expressions and may have no obvious place in the source code to point to. You could probably find a place to store it in all of the various things that can explicitly declare a type but it's definitely not straightforward to do that and would need some work, and nothing else in the compiler has needed it.

Your approach may work for your use case, I don't know, but it's not comprehensive. Your type can depend on an arbitrarily complex chain of constant function calls that can reference parameters within them, or can rely on the types of other variables that themselves depend on parameters.

MikePopoloski, 2025-09-06:
I'd be open to something like extending the `ASTContext::evalDimension` function to return the expression(s) that were involved with binding the dimension, which would make it more straightforward to implement the approach I first suggested. Feel free to open a PR if you want that. Otherwise I'm going to assume you've figured out a different workaround that works for your use case.

closed 2025-09-06 by MikePopoloski

pieter3d, 2025-09-06:
I assume you mean a feature request? 
Yes for now I've used my above mentioned workaround which is certainly not foolproof.