MikePopoloski/slang #1507: `--depfile-trim` failed to remove empty dummy file

donn, 2025-09-09:
**Is your bug report related to the C++ slang project or the pyslang Python bindings?**
C++ slang

**Describe the bug**
For a super simple test case here, dummy.v, a file that should not be counted as a dependency, is printed to `--all-deps` even though `--depfile-trim` is included. 

There are no symbol dependencies between dummy.v and the top module specified.

**To Reproduce**
Build slang and:

```
#!/usr/bin/env bash
set -euo pipefail

mkdir -p includes

cat > includes/macros.vh <<'EOF'
`define MODULE_B_NAME module_b
EOF

cat > dummy.v <<'EOF'
module dummy();

endmodule
EOF

cat > module_b.v <<'EOF'
`include "macros.vh"

module `MODULE_B_NAME (
    input wire clk,
    input wire in_b,
    output wire out_b
);
    assign out_b = in_b;
endmodule
EOF

cat > top.v <<'EOF'
`include "macros.vh"

module top_module (
    input wire clk,
    input wire in_b,
    output wire out_b
);
  `MODULE_B_NAME impl (.clk, .in_b, .out_b);
endmodule
EOF

cat > list.f <<'EOF'
+incdir+includes
module_b.v
top.v
dummy.v
EOF

./build/bin/slang -C list.f --all-deps out.f --depfile-trim --depfile-sort --top top_module

if grep -q dummy.v out.f ; then
    echo "dummy not trimmed"
    exit -1
else
    echo "ok!"
fi

```

**Additional context**
N/A

MikePopoloski, 2025-09-14:
This is because you are forcing all of your input files to be combined together into a single compilation unit via `-C`. If you allow them to be treated as independent units by using `-F` then they become separable and the dummy file will be trimmed out. Trimming can only happen at the compilation-unit level currently; it is assumed that such units are indivisible.

MikePopoloski, 2025-09-21:
Closing since there's been no followup.

closed 2025-09-21 by MikePopoloski