MikePopoloski/slang #1515: segfaults when using --lint-only

jcurtiss8086, 2025-09-13:
While working on Slang bringup for a larger repo, I wanted to try ```--lint-only``` to see if I could get that passing as a first step, and got a segfault.  I went back and added ```--lint-only``` to previous repos where I enabled Slang and also got segfaults.   I haven't yet found a small reproducer, but will update this issue with one.

Here's the output from a debug build.
```
terminate called after throwing an instance of 'slang::assert::AssertionException'
  what():  Assertion 'blocks.empty()' failed
  in file slang/source/ast/Statement.cpp, line 65
  function: slang::ast::Statement::StatementContext::~StatementContext()
```

jcurtiss8086, 2025-09-13:
The UVM library can reproduce the segfault:
```
slang --lint-only -I $UVM_HOME/src $UVM_HOME/src/uvm_pkg.sv
```

MikePopoloski, 2025-09-13:
When I tried this locally with UVM I got a different assertion when running analysis. Analysis should not run in lint-only mode so I fixed that here: ce920bac9fb4b9258142b46f88efaf315140d287

Not sure if that fixes the underlying issue for you, but after that UVM passes cleanly with --lint-only.

jcurtiss8086, 2025-09-13:
Yes, that solved it for all of the repos. Thanks!

closed 2025-09-13 by jcurtiss8086