MikePopoloski/slang #1521: SyntaxRewriter removes endif directive when replacing a subsequent statement

cyril0124, 2025-09-19:
# Description
The following is a code snippet used to reproduce this issue:
```c++
#include "fmt/core.h"
#include "slang/driver/Driver.h"
#include "slang/syntax/SyntaxPrinter.h"
#include "slang/syntax/SyntaxTree.h"
#include "slang/syntax/SyntaxVisitor.h"

int main(int argc, char **argv) {
    auto tree = slang::syntax::SyntaxTree::fromText(R"(
module test(
    input wire clock,
    input wire reset,
    output reg [3:0] o_signal
);
wire t;
`ifndef SYNTHESIS
initial begin
    $display("hello");
end
`endif
assign t = o_signal[1];

endmodule
)");

    struct Rewriter : public slang::syntax::SyntaxRewriter<Rewriter> {
        void handle(const slang::syntax::ContinuousAssignSyntax &syntax) {
            replace(syntax, parse("assign t = 2;"));
        }
    };

    auto newTree = Rewriter().transform(tree);
    fmt::println("{}\n", slang::syntax::SyntaxPrinter::printFile(*newTree));
    return 0;
}
```

# Expected Behavior
The SyntaxRewriter should only replace the assign statement and preserve the unrelated `endif directive. The resulting code should be:
```verilog
module test(
    input wire clock,
    input wire reset,
    output reg [3:0] o_signal
);
wire t;
`ifndef SYNTHESIS
initial begin
    $display("hello");
end
`endif
assign t = 2;

endmodule
```

# Actual Behavior
The SyntaxRewriter incorrectly removes the `endif directive along with the original assign statement. The resulting code is syntactically invalid:
```verilog
module test(
    input wire clock,
    input wire reset,
    output reg [3:0] o_signal
);
wire t;
`ifndef SYNTHESIS
initial begin
    $display("hello");
endassign t = 2;

endmodule
```

MikePopoloski, 2025-09-19:
Trivia (which includes preprocessor directives) is always associated with the next token in the stream. Perhaps there should be some helper method that replaces a node but preserves the leading trivia of the first token, but it's not super difficult to write such a method yourself.

cyril0124, 2025-09-20:
> Trivia (which includes preprocessor directives) is always associated with the next token in the stream. Perhaps there should be some helper method that replaces a node but preserves the leading trivia of the first token, but it's not super difficult to write such a method yourself.

Hi Mike, after several attempts, I found that the following code can preserve trivia:
```verilog
    struct Rewriter : public slang::syntax::SyntaxRewriter<Rewriter> {
        void handle(const slang::syntax::ContinuousAssignSyntax &syntax) {
            auto newNode = &parse("assign t = 2;");
            auto tk      = newNode->getFirstTokenPtr();
            *tk          = tk->withTrivia(alloc, syntax.assign.trivia());
            replace(syntax, *newNode);
        }
    };
```

I believe other users might run into this issue as well, perhaps we could add a parameter like `preserveTrivia` to SyntaxRewriter’s `replace` method, defaulting to `false`.

MikePopoloski, 2025-09-21:
Yes, it's likely that such a helper would be useful for others too.

cyril0124, 2025-09-21:
Also, other helper methods of `SyntaxRewriter` can also provide similar parameters, not just `replace`, such as `insertAfter` etc.

> > Trivia (which includes preprocessor directives) is always associated with the next token in the stream. Perhaps there should be some helper method that replaces a node but preserves the leading trivia of the first token, but it's not super difficult to write such a method yourself.
> 
> Hi Mike, after several attempts, I found that the following code can preserve trivia:
> 
>     struct Rewriter : public slang::syntax::SyntaxRewriter<Rewriter> {
>         void handle(const slang::syntax::ContinuousAssignSyntax &syntax) {
>             auto newNode = &parse("assign t = 2;");
>             auto tk      = newNode->getFirstTokenPtr();
>             *tk          = tk->withTrivia(alloc, syntax.assign.trivia());
>             replace(syntax, *newNode);
>         }
>     };
> I believe other users might run into this issue as well, perhaps we could add a parameter like `preserveTrivia` to SyntaxRewriter’s `replace` method, defaulting to `false`.

closed 2025-11-06 by MikePopoloski