MikePopoloski/slang #1524: Ambiguous class usage not flagged as an error

jcurtiss8086, 2025-09-22:
This is another case where VCS and Slang accept the code but Questa flags an error.

The class ```C``` exists in two different packages.   The 2nd package imports the first, and so the usage of ```C``` in package ```B``` becomes ambiguous as to which definition of ```C``` should be used.   

Should an error be flagged at either the definition of ```C``` within package ```B```, or at the usage of ```C``` within class ```D```?

```Systemverilog
package A;

    class C;
    endclass

endpackage

package B;
    import A::*;

    class C;
    endclass

    class D;
        C c;
    endclass

endpackage
```

MikePopoloski, 2025-09-22:
There is no ambiguity. 26.3 goes into depth on this topic, but the summary is that wildcard imports don't actually import anything right away; instead they make identifiers "potentially locally visible" and they only become imported if there is no matching local identifier declared in the scope.

I also tried your example with Questa 2024.3 and saw no error.

jcurtiss8086, 2025-09-22:
Thanks -- considering this to be a Questa bug.   I'm also having trouble reproducing with Questa unless I run with the full codebase.

closed 2025-09-22 by jcurtiss8086